devbisme/kinet2pcb #1: Netlist files don

ScottHilton, 2019-10-31:
* kinet2pcb version:
* Python version:
* Operating System:

### Description

Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.

### What I Did

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here.
```

xesscorp, 2019-11-01:
Hi, Scott. My apologies for the difficulties with `kinet2pcb`. As I always say: "it can never be easy."

To examine the problem, I created a small design with a resistor, capacitor, and LED from the `device.lib` symbol library. Inside EESCHEMA, I ran the `cvpcb` tool to assign footprints to the symbols, and then I exported a `.net` file.

Inside the `.net` file, I had three components that all looked similar to this:

```
  (components
    (comp (ref C1)
      (value C)
      (footprint Capacitor_SMD:CP_Elec_5x3)
      (datasheet ~)
      (libsource (lib Device) (part C) (description "Unpolarized capacitor"))
      (sheetpath (names /) (tstamps /))
      (tstamp 5DBC3998))
    (comp (ref D1)
      (value LED)
      (footprint LED_SMD:LED_0201_0603Metric)
      (datasheet ~)
      (libsource (lib Device) (part LED) (description "Light emitting diode"))
      (sheetpath (names /) (tstamps /))
      (tstamp 5DBC3E23))
    (comp (ref R1)
      (value R)
      (footprint Resistor_SMD:R_0603_1608Metric)
      (datasheet ~)
      (libsource (lib Device) (part R) (description Resistor))
      (sheetpath (names /) (tstamps /))
      (tstamp 5DBC51E3)))
```
As you can see, all the components have a footprint field that consists of the footprint library nickname and the footprint name separated by a colon.

You also need a `fp-lib-table` file that associates the footprint library nicknames with the locations of their  footprint directories. This file should be created when you installed KiCad and should be in the `\Users\<your name>\AppData\Roaming\KiCad` directory. A typical entry in that file looks like this:

```
(lib (name LED_SMD)(type KiCad)(uri ${KISYSMOD}/LED_SMD.pretty)(options "")(descr "Light emitting diodes (LED), surface mount"))
```

So my questions are:
1. Why did KiCad give you get a `.net` file with just the footprint name and no library nickname or colon?
2. Do you have an `fp-lib-table` and is it in the right place?
3. What version of KiCad are you running? (I'm using the latest release: 5.1.4 under Windows 10.)

xesscorp, 2019-11-25:
I haven't gotten any reply on this. Should I close this issue?

ScottHilton, 2019-11-25:
Sorry, I got pulled away to another project. I will eventually get back to this project. I think I should be able to fix it with the help you already gave. Thanks for the help.

closed 2019-11-25 by ScottHilton