devbisme/kinet2pcb #2: pcbnew.FootprintLoad do not find the requested footprint

NilujePerchut, 2019-11-02:
Hi,
   Please find below a problem encountered while trying to use kinet2pcb.
   I am pretty sure that I missed something important but I cannot find what.
   Can you give me some pointers to help me on my way ?
Thanks in advance,
Cédric

* kinet2pcb version: 0.1.0
* Python version: 2.7.12
* Operating System: Ubuntu 16.04

### Description

Using a Skidl generated netlist (attached to this request) I try to convert to my netlist to a PCB.
kinet2pcb crashes because pcbnew.FootprintLoad did not found the footprint (returned None).
kinet2pcb crashes on the first component found while:
     lib_uri = '${KISYSMOD}/Capacitor_SMD.pretty'
     fp_name= u'C_0603_1608Metric_Pad1.05x0.95mm_HandSolder'
which seems correct on my system.
KSYSMOD is exported before calling kinet2pcb

### What I Did
Inside a Makefile:
```
( export KIPRJMOD=/home/niluje/Work_local/keyboard/PCB/pcb_new/rcs && \
  export KSYSMOD="/usr/share/kicad/modules" && \
  python2 /home/niluje/Work_local/keyboard/PCB/pcb_new/tools/kinet2pcb.py -d7 -w -nb --input /home/niluje/Work_local/keyboard/PCB/pcb_new/build/kb.net  --output /home/niluje/Work_local/keyboard/PCB/pcb_new/build/kb_base.pcb )
Traceback (most recent call last):
  File "/home/niluje/Work_local/keyboard/PCB/pcb_new/tools/kinet2pcb.py", line 274, in <module>
    main()
  File "/home/niluje/Work_local/keyboard/PCB/pcb_new/tools/kinet2pcb.py", line 267, in main
    kinet2pcb(args.input, args.output)
  File "/home/niluje/Work_local/keyboard/PCB/pcb_new/tools/kinet2pcb.py", line 145, in kinet2pcb
    fp.SetParent(brd)
AttributeError: 'NoneType' object has no attribute 'SetParent'
```
PS: I get the same error using the pypi package.
PS2: I can provide you with the full project if you want

[kb.net.zip](https://github.com/xesscorp/kinet2pcb/files/3800919/kb.net.zip)

xesscorp, 2019-11-04:
I looked at your .net file. It looks OK to me.

In your Makefile, you export KSYSMOD, but further up you show a lib_uri using KISYSMOD. Is this the cause of the problem?

NilujePerchut, 2019-11-04:
Thanks a lot.  It is working now. This is as simple as forgetting an "I".
I am so sorry for the disturbance.
And thank you for this great tool (and for skidl ...)

closed 2019-11-04 by NilujePerchut