devbisme/kinet2pcb #5: kinet2pcb needs a 2to3 pass

sowbug, 2020-02-29:
```
Traceback (most recent call last):
  File "/home/miket/.local/bin/kinet2pcb", line 11, in <module>
    load_entry_point('kinet2pcb', 'console_scripts', 'kinet2pcb')()
  File "/home/miket/.local/lib/python3.6/site-packages/kinet2pcb-0.1.0-py3.6.egg/kinet2pcb/kinet2pcb.py", line 264, in main
    kinet2pcb(args.input, args.output)
  File "/home/miket/.local/lib/python3.6/site-packages/kinet2pcb-0.1.0-py3.6.egg/kinet2pcb/kinet2pcb.py", line 119, in kinet2pcb
    fp_libs = LibURIs(get_global_fp_lib_table_dir(), ".")
  File "/home/miket/.local/lib/python3.6/site-packages/kinet2pcb-0.1.0-py3.6.egg/kinet2pcb/kinet2pcb.py", line 57, in __init__
    self.load(fp_lib_table_dir)
  File "/home/miket/.local/lib/python3.6/site-packages/kinet2pcb-0.1.0-py3.6.egg/kinet2pcb/kinet2pcb.py", line 106, in load
    uri = rmv_quotes(uri)
  File "/home/miket/.local/lib/python3.6/site-packages/kinet2pcb-0.1.0-py3.6.egg/kinet2pcb/kinet2pcb.py", line 21, in rmv_quotes
    if not isinstance(s, basestring):
```

Running `2to3-2.7 -w /home/miket/.local/lib/python3.6/site-packages/kinet2pcb-0.1.0-py3.6.egg/kinet2pcb/kinet2pcb.py` changes this one line:

```
+++ /home/miket/.local/lib/python3.6/site-packages/kinet2pcb-0.1.0-py3.6.egg/kinet2pcb/kinet2pcb.py	(refactored)
@@ -18,7 +18,7 @@
 
 def rmv_quotes(s):
     """Remove starting and ending quotes from a string."""
-    if not isinstance(s, basestring):
+    if not isinstance(s, str):
         return s
```

Would you please make that change and roll out 0.1.1 or 0.2.0? Thanks!

xesscorp, 2020-03-04:
I think I've fixed this. Give it a try and let me know and then I'll release it on Pypi.

`pip install git+https://github.com/xesscorp/kinet2pcb`

sowbug, 2020-03-05:
That did the trick. Thank you.

xesscorp, 2020-03-09:
Good. Latest release is now on Pypi.

closed 2020-03-09 by xesscorp