devbisme/kinet2pcb #6: Is there a way to place all the components in netlist file genrated from skidl to place like the kicad netlist importer do?

HTerminal, 2020-07-12:
Hello 
I am working on a mixed project combined skidl and kinet2pcb.

I want to import netlist generated from skidl to kicad pcb. I used kinet2pcb which worked fine.

The only problem is that it places all the components on 0,0 axis. Is there any way i can place them like the default kicad netlist importer do.

Thank you 

Regards
Hitesh

xesscorp, 2020-07-12:
PCBNEW already includes the ability to do this:

1. Define a board outline. (**This is important!** The tools need a board area in which to place the components.)
2. Select all the components.
3. Click on Place => Place Footprints Automatically  => Automatically Place Selected Components

HTerminal, 2020-07-12:
Hello @xesscorp 
Thank you for the quick reply.

Is it GUI based?

Because i need to do this all from python script.
if it is possible from python can you share any piece of code.

Below i can explain my steps.
1. I used skidl to define my circuit and then exported the netlist (however i have some questions regarding skidl too, which i ll share on skidl repository) 
2. Now I installed **kinet2pcb** in python3 form pip3.
3. I used this command `kinet2pcb -i my_project.net` which gave `my_project.kicad_pcb`. 
4. I opened the pcb file which gave all the components on (0,0) location.
Now my question is, can i without opening pcbnew arrange all components without overlapping them just by using python?

Thank You

Regards
Hitesh

devbisme, 2022-11-22:
The `kinet2pcb` tool now includes a call to `hierplace.hier_place()` to careat a non-overlapping arrangement of footprints that mirrors the hierarchy of the SKiDL source code.

closed 2022-11-22 by devbisme